ut-robotics/picr21-team-shaq #19: Hardcoded serial port

Akustav, 2021-11-24:
https://github.com/ut-robotics/picr21-team-shaq/blob/8ae20e5501d61d2b16a69e7d05405c0ee861823d/src/comm.py#L22

The serial port is currently hard coded but can change when the device experiences connectivity issues or for other reasons.

Use [serial.tools.list_ports](https://pyserial.readthedocs.io/en/latest/tools.html) and port info to find the correct serial port.

NikitaNaumov98, 2021-12-06:
I get "module 'serial' has no attribute 'tools'" when trying to use this (but it works when ran from terminal). I think it's only supposed to be used in terminal, not in code.

ReikoR, 2022-01-12:
https://github.com/ut-robotics/picr21-team-shaq/blob/d2f5e877dab3a3672382eb2da5fde1242d83384e/src/comm.py#L12

This might not work if there are more serial port devices connected. 

Parameters from [serial.tools.list_ports.ListPortInfo](https://pyserial.readthedocs.io/en/latest/tools.html#serial.tools.list_ports.ListPortInfo) should be used to find the correct device.

closed 2022-01-16 by ReikoR