ut-robotics/picr21-team-shaq #29: Implement proper state machine and implement proportional diving

Akustav, 2021-11-24:
https://github.com/ut-robotics/picr21-team-shaq/blob/8ae20e5501d61d2b16a69e7d05405c0ee861823d/src/main.py#L46-L76

The current logic is very hard to maintain since everything is written in one huge block. It is difficult to implement additional logic like aiming or throwing. Example:

```
from enum import Enum

class State(Enum):
 FIND = 0
 DRIVE = 1
 AIM = 2
 THROW = 3

def handleFind(coordinates):
    print("finding_" + str(coordinates))

def handleDrive(coordinates):
    print("driving_" + str(coordinates))
    
ballData = ["someData"]

state = State.FIND

# main loop that executes some logic
while True:
  if state == State.FIND:
    # do find logic
  elif state == State.DRIVE:
    # do drive logic
  else:
    #handle error since no state is implemented for current state being handled
```

Also, the driving is using constants. It should use some form of proportional driving. An example picture with some explanations is below:

[image]

Tsapu, 2022-01-18:
State machine is present. Ball approach now uses proportional speed.

ReikoR, 2022-01-19:
https://github.com/ut-robotics/picr21-team-shaq/blob/3b1b4c18f1ef5eceefa19bf141379b7d1e02c075/src/main.py#L78-L85

https://digilabor.ut.ee/index.php/2021_PiCR_grading_and_tasks#Robot_follows_a_ball
> Robot must follow the ball by simultaneously rotating and moving towards it.

Currently robot is not moving and rotating at the same time.

I recommend cleaning up the movement code a bit and having something similar:

```python
def omni_components(self, speed, robot_angle, rotation_speed):
    return [
        self.wheelLinearVelocity(speed, 120, robot_angle) + rotation_speed, # Wheel 1 (right)
	self.wheelLinearVelocity(speed, 240, robot_angle) + rotation_speed, # Wheel 2 (left)
	self.wheelLinearVelocity(speed, 0, robot_angle) + rotation_speed,   # Wheel 3 (back)
    ]

def move_omni(self, speed, robot_angle, rotation_speed):
    omni_components = self.omni_components(speed, robot_angle, rotation_speed)
    self.sendSpeed(omni_components)
    
def move_omni_xy(self, speed_x, speed_y, rotation_speed):
    robot_angle = math.atan2(speed_y, speed_x)
    speed = math.hypot(speed_x, speed_y)
    self.move_omni(self, speed, robot_angle, rotation_speed)
```

Most of the movements (if not all) could probably be done with `move_omni_xy` function.

When moving towards a ball, `speed_y` could be proportional the distance to ball and `rotation_speed` could be proportional to  the ball x coordinate offset from the frame center.

closed 2022-01-22 by Akustav