ut-robotics/picr21-team-shaq #54: Move game state handling into separate functions

ReikoR, 2022-01-16:
https://github.com/ut-robotics/picr21-team-shaq/blob/4a55a202eb090c13c86e718c11b7f69319990997/src/main.py#L57-L61
https://github.com/ut-robotics/picr21-team-shaq/blob/4a55a202eb090c13c86e718c11b7f69319990997/src/main.py#L87-L90

Shortening the code would help a bit with readability.

Tsapu, 2022-01-18:
Will do this for final code, for now it's much easier to develop if we can oversee the whole game logic sequentially and change things on a whim easily.

Akustav, 2022-01-22:
To me the current project structure does not seem to be easy to modify. All states handle motion in a separate class that requires 2 files already. If you wanted to add different motions for different states, you need to create separate method in the moveControl since most methods used in the main are very specific, like case_ball(), center_ball() or align_for_throw(). It feels like half of the behavior logic has leaked into Movement.py.